AEFeinstein/Super-2024-Swadge-FW #442: Cosplay Crunch

jarettmillard, 2025-05-05:
## Summary
A fast-paced cosplay-themed microgame collection a la Warioware. A microgame typically consists of successfully performing a single action within 5-ish seconds. Microgames are crafting themed: "Sew", "Paint", "Try not to cry", "Glue", etc. Every failed microgame reduces your lives ("Days until MAG") by 1 until the fated day is upon you.

## Technical Spec
Individual microgames will be implemented as a struct with callbacks, similar to how modes are implemented, only with fewer callbacks and a few extra data fields. The game mode will randomly pick a microgame, delegate calls to it for init, game logic and drawing, and teardown, and determine if the user was successful or not based on a timeout and feedback from the microgame.

Ideally the mode itself will handle as much logic as possible and keep the microgames themselves slim. For example, timing the games should happen in the main game mode, allowing for less duplicated logic and a more consistent timer UI across different microgames.

Swadgepass can potentially be utilized to share high scores with other users.

Multiplayer is planned, either hotseat or networked.

## UI
Title screen menu:
- Play single player
- Play multiplayer
- View high scores

The game itself consists of a rapid-fire series of short games, with an inevitable game over screen.

## Mockups
Nothing yet

## How to Test
Play it I guess?

AEFeinstein, 2025-05-05:
Ooh, I like the vibe and I like the general architecture. If you want to do a proof-of-concept before getting too deep into game design or theme, you could make a set of microgames which are just "press up," "press down," "press left," and "press right."

Actually typing that out, there are _slight_ similarities to Swadge It (#441), but Cosplay Crunch seems way more fun anyway.

AEFeinstein, 2025-05-05:
@emilyanthony4244 probably knows some Cosplay verbs

AEFeinstein, 2025-05-06:
@jarettmillard do you have a fork of the firmware you're working on or something?

jarettmillard, 2025-05-06:
> [@jarettmillard](https://github.com/jarettmillard) do you have a fork of the firmware you're working on or something?

No, I'm just working locally for now. I figured I'd work that out once I had something worth making a PR out of.

AEFeinstein, 2025-05-06:
My personal preference, and recommendation, is too commit early and often. Backing up data is generally good, and having the ability to backtrack if/when something breaks that had been working is really nice too. I'll add you as a collaborator here.

AEFeinstein, 2025-12-15:
Are there any outstanding changes or can this be closed out?

jarettmillard, 2025-12-15:
> Are there any outstanding changes or can this be closed out?

Only waiting on menu BGM.

AEFeinstein, 2025-12-15:
Cool. When you get it please add it to the jukebox too.

closed 2025-12-18 by jarettmillard