AEFeinstein/Super-2024-Swadge-FW #455: Port PiCross

AEFeinstein, 2025-05-31:
## Summary

Remember PiCross? It's back!

## Technical Spec

Basically the same game as before, but using current Swadge APIs. Will eventually get new puzzles

## UI

Standard Swadge menu

## Mockups

n/a, go play the version from 2023

## How to Test

Clear all puzzles, including the final one.

AEFeinstein, 2025-10-28:
@hunterdyar, @DebrisHauler, aside from Trophies, is there anything else to do before we call Pi-Cross 2 done?

AEFeinstein, 2025-10-28:
Ah yes, Magcon feedback, collected:
- [ ] The taller and wider puzzles are tough to see on the small screen. Is there a way to zoom in?
- [ ] Would be cool to have optional ambient looping music in the background.
- [ ] Just needs a few more puzzles. (Dac _always_ wants more)
- [ ] There should be a way once you solve the puzzle to go straight into the next puzzle once you solve the current one. Or at least put the cursor on the next puzzle when you go back to the list.
- [ ] When you complete a puzzle and it's animating into place before the color is filled in, it looks weird if the puzzle has mostly black around the edges. Maybe a 1px white or grey border or making the background dark grey instead of black would make it look better.

I'd treat those as suggestions, not hard to-do. @DebrisHauler can you check those off as you do (or chose not to do) them? I'm not sure how github works with checkboxes across accounts.

Also praise
* Excellent overall.
* LOVE me some picross.

DebrisHauler, 2025-10-29:
One last thing in picross-feedback branch. That is music. I'm not doing zoom or more puzzles.

AEFeinstein, 2025-10-29:
Oh good, you can click the checkboxes

closed 2025-11-27 by AEFeinstein