AEFeinstein/Super-2024-Swadge-FW #470: Add sleep functionality to the ch32v003 library

cnlohr, 2025-07-19:
swadge_matrix.h contains a few functions for working with the LED matrix display.  It would save a little power if we could put the ch32v003 to sleep after the data is written to the DMA engine, and the DMA engine will continue to scan out the LEDs.

But this will provide a vast amount of processor saving when running the emulator! So at some point this needs to be addressed.

AEFeinstein, 2025-07-19:
I do like not having the emulator hog my CPU!

AEFeinstein, 2025-10-25:
@cnlohr found this older ticket and it's freshly relevant. The ch32is drawing ~7mA, and I would love to get it to sleep. Ideally the ch32 is slept here:
```c
// Check if it should be blanked
danceState->buttonPressedTimer += elapsedUs;
if (danceState->buttonPressedTimer >= TFT_TIMEOUT_US)
{
    disableTFTBacklight();
    danceState->blankScreen = true;
}
```
And then woken back up here:
```c
// If a button has been pressed recently
if (danceState->buttonPressedTimer < TFT_TIMEOUT_US)
{
    // Turn the screen on
    enableTFTBacklight();
    setTFTBacklightBrightness(getTftBrightnessSetting());
    danceState->blankScreen = false;
    // Draw to it
    drawMenuMega(danceState->menu, danceState->menuRenderer, elapsedUs);
}
```
I'll take whatever current savings I can get!