AEFeinstein/Super-2024-Swadge-FW #483: Swadgetamatone

jarettmillard, 2025-09-08:
## Summary
Annoying synth instrument with a sound approximating an otamatone.

## Technical Spec
Touch pad X adjusts pitch.
Touch pad Y adjusts volume.
D-pad left/up/right play notes in the low/mid/high octaves.

## UI
The screen displays a mouth to match the touchpad movements.

## Mockups

[image]
[image]

## How to Test
Play the mode, ideally on actual hardware.

AEFeinstein, 2025-10-28:
@jarettmillard I think this is done? If there's nothing else you want, close it out please!

jarettmillard, 2025-10-28:
I still want to do one final adjustment to the eyes once I get test hardware.

closed 2025-11-13 by AEFeinstein